Dvogonen/cyclop_plus #3: Battery voltage mod + changes

CurtisFissel, 2016-07-02:
I just got done doing a slight mod to my Cyclops Goggles. 
[image]

Added a voltage divider to A2 pin (A1 seems to be mislabeled? It actually goes to A3 so i just decided to use the correctly labeled A2) 

I'm using a 10K and 4.7K resistor for the voltage divider. But now that I realize this is a 3.3v based system that won't be good enough for 3s battery sensing. The resistor values are currently configurable in code I have written so anyone can choose a suitable divider for their application.

Open to suggestions of standardizing the resistor values to make it more noob friendly and reduce code complexity.

Still working on the code but figured I would open the discussion

Curtis

CurtisFissel, 2016-07-03:
Did some work and added it to the main Channel Screen with a nice little icon and bar indicator.
[image]

Doesn't currently support 3s voltages(extra "1" digit) yet and the code is a little messy so im trying to fix that up.

Dvogonen, 2016-07-07:
Thats looking really nice.
I think that the best idea is to specify exactly which resistance values to use. If someone feels confident to start changing those, they will probably have no problems changing the code.
To be able to use 3s, 2s as well as 9 volts batteries  I think that a 3 to 1 divider is optimal. A 3s battery will have a maximum voltage of around 12,6 volts 12,6/4 = 3.15. Very close to 3.3.
The standard cookbook rule is to use  a resistance of 10K for analog inputs. But I guess that that is intended for 5 volt sources. With a 3s battery we should use at least 30K. How about using a  22K and a 68K resistor? Standard values and with safety margin. It also makes it possible to use replace the 68K resistor with three 22K ones in series.

MrChzl, 2016-07-12:
@Dvogonen  I'm going to assume that your suggestion for a 3 to 1 divider using a 22k and 68k resistor would work for up to the 3s battery you stated. However, would you recommend using 30k and 90k (or as close to those values as possible) if I plan to use a 3s battery at any point in time? @CurtisFissel I see you have a white wire coming from pin A2. Is that connected to Vout of your voltage divider? Also is Vin and Ground coming from positive and negative power to the receiver? @Dvogonen I've flashed my goggles with v1.3 and i absolutely love the layout! However in the code you state to not waste time if the battery indicator did not show up and on mine it does, however at around 8.1v from a 2s battery, the meter shows full charge. Would this indicate the need for the voltage divider to properly display battery charge state? I look forward to hearing back from both of you!

Dvogonen, 2016-07-15:
It turns out A1 (the real pin, not the mislabeled pad on the PCB)  is already wired so that it is possible to use it to read battery voltage. No divider is needed. The resolution is not optimal, but it works for a crude battery meter.  I have not heard from CurtisFissel in a while, so I am completing the code on my own.
I have committed the code in the 1.3 branch so this issue is closed

closed 2016-07-15 by Dvogonen

CurtisFissel, 2016-07-15:
Hey sorry about that. Was crazy busy this week and just haven't had anytime to work on this.

Here is my code for the 2s battery and Beeper setup. 
[cyclop_plus.zip](https://github.com/Dvogonen/cyclop_plus/files/365870/cyclop_plus.zip)

Feel free to grab anything from it. I assume your code will be much cleaner though. 

Curtis