Dvogonen/cyclop_plus #13: Cant change channels.

Cr381v3, 2016-09-02:
I got it all flashed and working. However it doesn’t seem to change channels. The frequency changes on the little display but not on the screen i seem to be locked into c1. I have an eachine racer 250 and was able to use it to scan through and connect and the OSD from the quad says voltage flight time and channel comes up as c1. If i use the Frequency scanner with waveform it looks like im broadcasting on every channel. But im not because i had to manually go thru each channel on the quad end to find what channel the reciver was on. Im confused

I can send pictures or video if you need to see it

Cr381v3, 2016-09-02:
Im Using ver. cyclops+ 1.4

jnowak1054, 2016-09-02:
When scanning the frequency, how close was the RX to the TX. Looks like the Eachine racer 250 comes with a 600mw transmitter which is quite powerful.  Try separating the quad and the cyclops by some distance, say 30-50 feet and see what the frequency scanner looks like then.

Cr381v3, 2016-09-02:
Good point. I didnt even think about that. Ill give it a shot after work
and let you know.

Thanks for taking the time.

Cr381v3, 2016-09-02:
Same issue. I even downgraded to 1.3 ..

Here is a video

https://youtu.be/OXGJ3QnLZZY

Cr381v3, 2016-09-02:
I used two different quads on different channels. And it auto locks on to
what ever drone is closest if both are on when i connect power to the
Cyclops. And auto changes to one or the other when powered up depending on
which quad is powered. Still not able to change channels. And i don't think
the screen displays the correct channel im on.   Still really confused..

jharmn, 2016-09-02:
I'm seeing similar issues on my goggles. Usually the first few times I connect, I get either full signal across the band, or no signal across the band. If I connect/scan/disconnect battery a few times, it will always work after 3 or 4 tries. 

My assumption is this is a bad rx5808-pro receiver chip that needs to be replaced (once it's warmed up, it's near 100% reliable). I was just going to order a replacement and wait for it to die completely.
Could be software, but I'm guessing it's just hit or miss Chinese hardware as usual.

Cr381v3, 2016-09-03:
Yea its a shame. Not bad for $50. They still connect just not as
advertised. Not a big deal. I was looking at swapping out the reciver to
the eachine ER32 if it gives me any more trouble. Thanks  lfor the help.

Dvogonen, 2016-10-05:
I missed this error report. I guess that the owner has solved his problem by now, but I am answering for the benefit of others. 

I checked your video and as far as I can see your receiver module is not taking any notice whatsoever about the commands sent by the processor. The processor does not get information about the current channel from the receiver, it assumes that the receiver obeys the commands given and displays whatever value it has sent. 

What is strange is that both your processor and receiver module work just fine. This is clear from the video. The RSSI value is also read correctly, which means that the connection on that pin is intact. 

My guess is that the problem is the SPI communication between the processor and the receiver. If you are lucky your problem depends on a faulty solder joint. The three leftmost solder points on the receiver are the SPI interface. Resolder these. This is an easy solder job. If this doesn't work, trace the traces back to the processor and resolder the corresponding legs. This is anything but easy.

Dvogonen, 2017-01-29:
Closing this one since no new information has emerged for a long time.

closed 2017-01-29 by Dvogonen