Dvogonen/cyclop_plus #15: Battery alarm recalibration

Dvogonen, 2016-10-12:
The battery alarm sets in to early according to this post by awap at RCGroups:
"Hi, I'm very happy with version 1.4 and the Eachine DVR installed. One whish however: I'm on a 3S battery and the beeper starts at 11.4V (3.8V per cel) and the second level beep starts at 11,2V (3,73V per cel). This is too soon. Is there a way to start the beeper at 3x3.65= 10.95 -11.0 V?"

The behavior is not really a bug, the level was set unnecessarily low.

knalltuete, 2016-10-27:
I use a modded v1.5 version and also had to adapt the values for the alert levels to fit. It seems the values depend on each single goggles device and might vary in a wider range.

RANDY45, 2017-02-17:
I can't get the battery alarm to work am I doing something wrong.

RANDY45, 2017-02-17:
Email me at charlesphillips1972@gmail.com 
Dvogonen contact me plz I had downloaded the 1.5 version and two problems I have 3s battery don't work and buzzer

Dvogonen, 2017-03-06:
Version 1.5 is released and has got a fixes for the issue with too high alarm limits. It has also got a configuration option for the voltage level.

closed 2017-03-06 by Dvogonen