Dvogonen/cyclop_plus #18: OLED not always initialised on power up

UncleShark, 2016-11-26:
I have noticed that the OLED does not always initialise on power up. once it fails it can take 4 or 5 power cycles for it to wake up. Have tried 2 OLED from different vendors. Is this a hardware fault?

paulius2k, 2016-12-21:
I can confirm this issue.
Powering the receiver board via USBASP will **always** light up the OLED. However it can take sometimes up to 10 cycles to light up the OLED with a LiPo.

Dvogonen, 2017-01-29:
The OLED is initialized from the CPU via the I2C buss, what you are seeing is the this initialization fails. The initialization only works if the OLED has already booted properly when the power was applied. A to high or to low voltage might cause the OLED not to boot. The I2C communication is one way only, so the CPU has no way of knowing if the initialization worked.

I had this kind of problem during development with an OLED unit that had a faulty controller. I changed the timing of the initialization and tried multiple initializations, but that had no effect. At the end I gave up and switched OLED, which solved the problem. My conclusion was that it was not the initialization from the CPU that failed, but the boot at power on. There is nothing to be done about that in software. The observation that power feeding via USB works, while power feed from a battery is sketchy, strengthens that conclusion.

Check the voltage feed to the OLED. It should absolutely not be above 5.5 volts. Anything between 3.3volts and 5volts ought to work, but sadly the OLEDs come from different manufacturers and have different tolerances.