Dvogonen/cyclop_plus #19: E3 and R1 are in inverse in frequency order when using "Next channel" function or Scan

Eulkomar, 2017-01-13:
Hello Dvogonen,
First of all thank you and congratulation for this so nice project ! 
It is so usefull to be sure on which frequency we are locked onto !!
Great Job and your user interface is great too !

I found one little bug and I am, for instance, unable to solve it by myself. (I did not understand yet what was the logic behing progmen index, progmem value and frequencies order.)

Here is the bug description : 
- I set my VTX to  R1.
- Making a scan, Cyclops+ locks on E3 (5665 MHz), despite the fact that R1 is a lower frequency (5658). 
  Not a big deal. 
- But when I want to use "previous frequency", to reach R1, I jump to E4 (5645).
- R1 appears in fact after E3

One can leave with this but on a such well done software it should be a pity not to correct it. 
It is looking like  two values that are inverted. 
I am convinced you will see this in a snap when checking your code... ;-)

Best regards
Marc

Eulkomar, 2017-01-13:
Hello again,
Checking again I finally saw my understanding with the logic of frequencies lists.
The typo in the code is here : 

/const uint8_t positions[] PROGMEM = {
  19, 18, 32, 17, 33, 16,  7, 34,  8, 24,  6,  9, 25,  5, 35, 10, 26,  4, 11, 27,
  3, 36, 12, 28,  2, 13, 29, 37,  1, 14, 30,  0, 15, 31, 38, 20, 21, 39, 22, 23
};
18 and 32 should be inverted.

I am registered on github only since few minutes, sorry if I made my remarks in a wrong way. 
I never made a program with arduino, now I am fighthing with it to compile this modification to test it :-)

Dvogonen, 2017-01-29:
Thank you for your input. I will check the issue. I appreciate you taking your time to register with Github to register the report. Although I do not check the report queue more than perhaps a couple of times per month, a filed error report will always be noticed

Dvogonen, 2017-03-06:
This issue has been fixed in the 1.5 release.

closed 2017-03-06 by Dvogonen