Dvogonen/cyclop_plus #20: Add L channels ??

mattyra, 2017-01-26:
Hello, Is it possible to add the L Channels for the band ??

https://github.com/sheaivey/rx5808-pro-diversity/issues/75

Dvogonen, 2017-01-29:
I just checked the datasheet for the RichWave receiver unit that is used by the Cyclops googles and according to it, the lowest possible frequency the receiver can receive is 5725 KHz. The L band consists of eight frequencies below 5725 (5362KHz-5621KHz). If the datasheet is correct, it is technically impossible to implement L-band support.
That said, harware can sometimes be used at out of spec settings. 5725 KHz is the lower end of the legal 5.8 MHz band, so it makes little sense for manufacturers to claim support for a wider frequency range, even if the hardware supports it. I am however not in a position to experiment with trying to use the receiver at L-band settings, since I do not have any transmitter that is L-band capable. So, for the moment the answer to the question is; 
No, it is not possible to implement L-band support.

mattyra, 2017-03-02:
Are you reopening this now ??

Dvogonen, 2017-03-02:
Yes, I have bought an L-Band capable transmitter and will test if it is possible to tweak the transmitter of the CYCLOPS to receive the lower frequencies. The specifications clearly says that it is not possible, but you never know.

mattyra, 2017-03-02:
Well thats great, let me know if you want the code tested..

Matt

Dvogonen, 2017-03-06:
I have just released 1.5 with full support for the L-band.

closed 2017-03-06 by Dvogonen